maiandrew23/LockBox #5: Testing Printer Subsystem

maiandrew23, 2021-02-21:
Test the printer for proper operation. 
1. Perform a UART Loopback test on the Raspberry Pi to ensure the serial port is operation. 
2. Make the necessary wire connections between the Pi and printer ensuring the proper voltage is supplied to the printer. 
3. Send setup commands to the printer if required.
4. Print out a test receipt on the printer and inspect for any issues.

DanielHunte, 2021-03-24:
### Perform a UART Loopback test on the Raspberry Pi to ensure the serial port is operation.

* Send AT command through serial

Expected Result: Receive a "OK" response

DanielHunte, 2021-03-24:
### Perform a UART Loopback test on the Raspberry Pi to ensure the serial port is operation.

The correct connections was made and the AT command was sent to the printer. No "OK" response was received but the printer printed "AT" instead. I then tried to print "Hello" which worked.
[image]
[image]

DanielHunte, 2021-03-24:
### Print diagnostic receipt

* Press the power button on the printer while connecting the printer to power.

Expected result: A diagnostic receipt should be printed

DanielHunte, 2021-03-24:
> ### Print diagnostic receipt
> * Press the power button on the printer while connecting the printer to power.
> 
> Expected result: A diagnostic receipt should be printed

When I do this the printer does not turn on (the led does not blink) and nothing is printed. I believe this is a power issue. Maybe not enough current is being supplied to the printer.

DanielHunte, 2021-03-24:
> When I do this the printer does not turn on (the led does not blink) and nothing is printed. I believe this is a power issue. Maybe not enough current is being supplied to the printer.

I tried this with a 5V 4A power supply directly connected to the printer and it still didn't work. After playing around with the printer I found out the correct way to print out the diagnostic sheet. At any point after the printer has turned on, hold down the button on the circuit board and the diagnostic sheet will print.
[image]
[image]

DanielHunte, 2021-03-24:
> When I do this the printer does not turn on (the led does not blink) and nothing is printed. I believe this is a power issue. Maybe not enough current is being supplied to the printer.

I connect the printer to the pi. With the pi on, only the first line of the diagnostic sheet prints. With the pi off, the first line is printed and the second line is printed very faintly. This confirms that the 5V 2A power supply does not supply enough current to the printer. Therefore the 5V 4A power supply will be used from now on. 
[image]

DanielHunte, 2021-03-24:
### Install relevant library and test printing different things

* Follow instructions from https://github.com/adafruit/Python-Thermal-Printer and install the relevant libraries.
* Run printertest.py 

Expected result: A receipt should be printed testing all the different font sizes, justifications, barcodes and images.

DanielHunte, 2021-03-24:
> ### Install relevant library and test printing different things
> * Follow instructions from https://github.com/adafruit/Python-Thermal-Printer and install the relevant libraries.
> * Run [...]

The 5V 4A power supply was used to power both the pi and the printer. A 2.1mm DC barrel jack was used to connect the power supply to the breadboard.

The instructions in the README.md were followed and printertest.py was run using python3 (NOTE: only works with python3 not python2).
[image]

commit 8b633365 (2021-03-31): Added printer library to directory and updated the demo and lockbox files, closes #5
closed 2021-03-31 by DanielHunte