maiandrew23/LockBox #6: Testing Solenoid Subsystem

maiandrew23, 2021-02-21:
To test the solenoid subsystem:
Set up the amplifier circuit using the provided TIP transistor with heat sink installed. Connect the output of the amplifier to the solenoid and apply Vcc to the amplifier circuit. Ensure the solenoid is operation in both ends and that the throw length is sufficient to reach the locking mechanism on the box chassis. Repeat a couple times to ensure it it is fully operational.

maiandrew23, 2021-03-19:
<h1>Solenoid Testing </h1>

- [ ] Set up Solenoid and driver circuit:
- [ ] Write solenoid setup and operation functions
- [ ] Update demo.py to test the solenoid

## Setting up driver circuit

1. Connect a 1.2k ohm resistor on the base of the TIP104 transistor.
2. Connect the transistor emitter to GND.
3. Connect one end of the solenoid to the collector and the other end to a 5v power supply.
4. Connect a diode in parallel with the solenoid with the anode (+) on the collector. 
5. Connect the other end of resistor to GPIO 16

## GPIO Setup
For the setup function, setup GPIO16 as an output. Write methods in the Solenoid class to extend and contract the solenoid.

## Testing Solenoid 
In demo.py, write a test script that will extend and contract the solenoid 5 times with 3 second pauses between contraction and extension. Make sure the circuit does not overhead during this process.

Expected Result: Solenoid should extend and contract instantly without any mechanical jams. The transistor and the Pi should not be hot to the touch after the test.

maiandrew23, 2021-03-24:
[image]

The solenoid circuit is set up on the bread board. The 1.2 ohm resistors are in between GPIO and the base of the transistor.
An LED is used to act as a diode, which is place in parallel with the solenoid.

maiandrew23, 2021-03-24:
[image]
lockbox.py
[image]
demo.py
The GPIO setup was written in lockbox.py which will write a HIGH for open and a LOW for closed.

maiandrew23, 2021-03-24:
The circuit did not work properly as expected. After troubleshooting, it was discovered that the solenoid was only drawing about 400 mA. 
[image]
To try the circuit without the transistor circuit, only 600 mA was measured. The minimum required current is 800 mA. 

Since its possible that there is not enough current provided out of the 5v pin on the pi, a different power source was used. 
[image]
In this setup, the solenoid is directly connected across a power adapter used for an IPAD with a rating of 5.2V 2.4A. The current did not get any higher. 

To increase the current as much as possible, the multimeter and jumper cables were removed so the power supply can be directly connected to the solenoid as shown.
[image]
In this setup, the solenoid had just enough force to activate. It was noticed that the usb and pi wires were very thin and could be the cause of the low current.

DanielHunte, 2021-04-26:
> [image]
> 
> The solenoid circuit is set up on the bread board. The 1.2 ohm resistors are in between GPIO and the base of the transistor.
> An LED is used to act as a diode, which is place in parallel with the solenoid.

Powering the solenoid with a 5V 4A supply, measured the voltage across the solenoid with the transistor connected and got 3.6V. This means the voltage across the transistor is 1.4V. The solenoid needs to be provided with at lease 5V to operate properly. We've determined that current is no longer an issue (using 4A supply) but voltage is. Therefore we will test the solenoid with a 9V boost converter.

DanielHunte, 2021-04-26:
### Boost Converter Test
Well will use a 9V Step-Up Voltage Regulator U3V12F9:
[image]

* Connect 5V power supply pin to Vin on the boost converter
* Connect power supply GND pin to GND on the boost converter.
* Measure the voltage across Vout on the boost converter and GND

Expected Result: Measure 9V on the multimeter

DanielHunte, 2021-04-26:
> ### Boost Converter Test
> Well will use a 9V Step-Up Voltage Regulator U3V12F9:
> [image]
>
> * Connect 5V power supply pin to Vin on the boost converter
> * Connect power supply GND pin to GND on the boost [...]

The mentioned connections were made and 9.09V was measured across the boost converter.:
[image]

DanielHunte, 2021-04-26:
### Test solenoid with transistor

* Connect one end of the solenoid to Vout on the boost converter
* Connect the other end of the solenoid to the collector of the transistor
* Connect the emitter of the transistor to GND
* Connect the base of the transistor to the one end of a 1k resistor
* Connect the other end of the resistor to GPIO16 on the pi
* Connect a diode across the solenoid is the opposite polarity
* Run demo.py and observe the solenoid

Expected result: The solenoid opens and closes at the specified times in demo.py

[image]

DanielHunte, 2021-04-26:
> ### Test solenoid with transistor
> * Connect one end of the solenoid to Vout on the boost converter
> * Connect the other end of the solenoid to the collector of the transistor
> * Connect the emitter of [...]

The mentioned connections were made and the solenoid open and closed as expected.
[image]
[image]

closed 2021-04-26 by DanielHunte