maiandrew23/LockBox #8: Testing Character LCD

DanielHunte, 2021-02-24:
- [ ] Test if it turns on
- [ ] Test backlight levels and set to a reasonable contrast
- [ ] Try setup commands and see if a correct response is received
- [ ] Write to both lines of the display
- [ ] Make sure response time is virtually zero

ffund, 2021-03-10:
### Test if it turns on 

Connect:

* Pin 1 (VSS) and pin 16 (BL- or K) are connected to the ground rail.
* Pin 2 (VDD) is connected to the 5V power rail.
* Pin 16 (BL+/A) is connected to the 5V power rail through a 220Ω series resistor.

Then, connect a GND pin from Pi to the GND rail, and a 5V pin from Pi to the 5V power rail. 

Expected result: nothing, until you have adjusted contrast.

ffund, 2021-03-10:
### Test contrast

Instructions: Turn the knob on the potentiometer to a middle position, to start. Connect
the outside pins of the potentiometer to the ground rail and 5V power rail, respectively. (It doesn’t matter
which of the two outside pins is at GND and which is at 5V - the trimpot is symmetric.) The middle pin of
the potentiometer is connected to LCD pin 3 (V0).

Carefully turn the knob on the trimpot until you see a row of white squares appear on your LCD display.

ffund, 2021-03-10:
### Try setup commands

Implement init, on, and off functions in library.

Test by running demo.py

Expected result: when the on function runs, the display turns on; when the off function runs, the display turns off.

maiandrew23, 2021-03-10:
> ### Test contrast
> Instructions: Turn the knob on the potentiometer to a middle position, to start. Connect
> the outside pins of the potentiometer to the ground rail and 5V power rail, respectively. (It doesn’t matter
> which of the two outside pins is at GND and which is at 5V - the trimpot is symmetric.) The middle pin of
> the potentiometer is connected to LCD pin 3 (V0).
> 
> Carefully turn the knob on the trimpot until you see a row of white squares appear on your LCD display.
> 
> Expected result:
> [image]

The potentiometer is connected to 5V, V0, and GND respectively. Bright squares were visible after turning the knob approximately 85% of the way.
[image]

maiandrew23, 2021-03-17:
> ### Try setup commands
> Implement init, on, and off functions in library.
> 
> Test by running demo.py
> 
> Expected result: when the on function runs, the display turns on; when the off function runs, the display turns off.

The init function was implemented Pins RS, E, and D4-D7 were set to GPIO.OUT.
A couple initial commands were sent to the LCD using write_arr_4bit function. 
[image]

When the display off function was called, the display was successfully turn off:
[image]

After running demo.py which initializes the display class and calls showText to display "Test1" on the LCD and finally printing "Display setup finished". 
Running demo.py was successful:
[image]
After running demo.py, "test1" is displayed on the LCD
[image]

closed 2021-03-17 by maiandrew23