maiandrew23/LockBox #9: Testing Keypad

DanielHunte, 2021-02-24:
- [ ] Test keypad using pi's internal pullup resistors
- [ ] If internal pullups are unreliable, test with external 10k resistors
- [ ] Make sure every individual button works
- [ ] Test how quickly buttons can be pressed in succession (testing ghosting)
- [ ] Enter a passcode and check if the output is correct

DanielHunte, 2021-03-17:
###Test continuity throughout keypad using multimeter

* Put multimeter in continuity mode and connect one probe to R1 and the other to R2 on the keypad.
* Press the corresponding button on the keypad and listen for beeping sound.
* Repeat for all other two pin combinations (R1, R2, R3, R4, C1, C2, C3).

Expected result: to hear the beeping noise for all pin combinations

DanielHunte, 2021-03-17:
> ### Test continuity throughout keypad using multimeter
> * Put multimeter in continuity mode and connect one probe to R1 and the other to R2 on the keypad.
> * Press the corresponding button on the [...]

The beeping sound was heard when pressing the relevant button for all pin combinations. The keypad functions properly.

DanielHunte, 2021-03-17:
### Test keypad using pi's internal pull up/down resistors

* Set GPIO17 to input and activate pull down resistor.
* Connect R1 to 3.3V pin on Pi
* Connect GPIO17 to one end of a 220 ohm resistor and connect the other end to C3.
* Write code to continuously read the input of GPIO17 and print whether button is pressed or not

Expected: When the "3" button is pressed, it prints "Button IS pressed" and when the "3" button is not pressed it prints "Button is NOT pressed

DanielHunte, 2021-03-17:
> ### Test keypad using pi's internal pull down resistors
> * Set GPIO17 to input and activate pull down resistor.
> * Connect R1 to 3.3V pin on Pi
> * Connect GPIO17 to one end of a 220 ohm resistor and connect the other end to C3.
> * Write code to continuously read the input of GPIO17 and print whether button is pressed or not
> 
> Expected: When the "3" button is pressed, it prints "Button IS pressed" and when the "3" button is not pressed it prints "Button is NOT pressed

The connections were executed as described.
[image]

The test worked as expected. When the "3" button was pressed, it prints "Button IS pressed" and when the "3" button is not pressed it prints "Button is NOT pressed". This verifies the functionality of the internal pull down resistor.
[image]

DanielHunte, 2021-03-18:
### If internal pullups are unreliable, test with external 10k resistors

* Press button multiple times
* Hold down one button for a period of 10 seconds

Expected: If internal pull down resistors are reliable, the keypad will still be functional

DanielHunte, 2021-03-18:
> ### If internal pullups are unreliable, test with external 10k resistors
> * Press button multiple times
> * Hold down button for a period of 10 seconds
> 
> Expected: If internal pull down resistors are reliable, the keypad will still be functional

Button was pressed multiple times and also held down for 10 seconds and the keypad was still functional after this.

DanielHunte, 2021-03-18:
### Make sure every individual button works

* Connect R1 to one end of a 220 ohm resistor and the other end of the resistor to GPIO2
* Connect R2 to one end of a 220 ohm resistor and the other end of the resistor to GPIO3
* Connect R3 to one end of a 220 ohm resistor and the other end of the resistor to GPIO17
* Connect R4 to one end of a 220 ohm resistor and the other end of the resistor to GPIO27
* Connect C1 to one end of a 220 ohm resistor and the other end of the resistor to GPIO22
* Connect C2 to one end of a 220 ohm resistor and the other end of the resistor to GPIO23
* Connect C3 to one end of a 220 ohm resistor and the other end of the resistor to GPIO24
* Write code to scan/cycle through rows and columns to obtain specific button pressed on keypad
* Press each individual button while running test (demo.py)

Expected result: When a button is pressed, the corresponding character is printed

DanielHunte, 2021-03-18:
> ### Make sure every individual button works
> * Connect R1 to one end of a 220 ohm resistor and the other end of the resistor to GPIO2
> * Connect R2 to one end of a 220 ohm resistor and the other end [...]

The connections mentioned were made:
[image]

The functionality for scanning the rows and columns was added to lockbox.py and the demo.py was updated to better test the keypad:
[image]

DanielHunte, 2021-03-18:
### Test how quickly buttons can be pressed in succession (testing ghosting)

* Press random buttons in succession (as an actually user would do) during demo
* Adjust the sleep time between button presses to alleviate ghosting

Expected result: All buttons pressed should be printed.

DanielHunte, 2021-03-18:
> ### Test how quickly buttons can be pressed in succession (testing ghosting)
> * Press random buttons in succession (as an actually user would do) during demo
> * Adjust the sleep time between button presses to alleviate ghosting
> 
> Expected result: All buttons pressed should be printed.

A sleep time of 0.2 seconds was sufficient to prevent ghosting. The keypad is functioning as expected.

closed 2021-03-18 by DanielHunte

maiandrew23, 2021-03-19:
#Solenoid Testing :smile: <h1> 

- [ ] Set up Solenoid and driver circuit:
- [ ] Write solenoid setup and operation functions
- [ ] Update demo.py to test the solenoid

## Setting up driver circuit <h2>

1. Connect a 1.2k ohm resistor on the base of the TIP104 transistor.
2. Connect the transistor emitter to GND.
3. Connect one end of the solenoid to the collector and the other end to a 5v power supply.
4. Connect a diode in parallel with the solenoid with the anode (+) on the collector. 
5. Connect the other end of resistor to GPIO 16

## GPIO Setup
For the setup function, setup GPIO16 as an output. Write methods in the Solenoid class to extend and contract the solenoid.

## Testing Solenoid 
In demo.py, write a test script that will extend and contract the solenoid 5 times with 3 second pauses between contraction and extension. Make sure the circuit does not overhead during this process.

Expected Result: Solenoid should extend and contract instantly without any mechanical jams. The transistor and the Pi should not be hot to the touch after the test.