maiandrew23/LockBox #21: Static assets (CSS etc) not loading in Flask UI

ffund, 2021-04-23:
When running the Flask app, static assets such as CSS files are not returned by the server. Note the 404s in the example below:

```
192.168.12.1 - - [23/Apr/2021 09:38:19] "GET / HTTP/1.1" 200 -
192.168.12.1 - - [23/Apr/2021 09:38:19] "GET /assets/bootstrap/css/bootstrap.min.css HTTP/1.1" 404 -
192.168.12.1 - - [23/Apr/2021 09:38:19] "GET /assets/css/Header-Dark.css HTTP/1.1" 404 -
192.168.12.1 - - [23/Apr/2021 09:38:19] "GET /assets/css/Navigation-Clean.css HTTP/1.1" 404 -
192.168.12.1 - - [23/Apr/2021 09:38:19] "GET /assets/css/styles.css HTTP/1.1" 404 -
192.168.12.1 - - [23/Apr/2021 09:38:19] "GET /favicon.ico HTTP/1.1" 404 -
```

ffund, 2021-04-23:
Try this directory structure:

* static HTML files are in `html`
* in `app.py`, set it up with `static_folder='html'` and then when sending static HTML files, just path their name, not `html/name`
* in HTML files, refer to static assets with path: `html/assets/...`

ffund, 2021-04-23:
See e.g. this example:

[image]

When testing, make sure to do a "hard refresh" (also refresh any cached items) after changing HTML or CSS - in Chrome, use Ctrl+Shift+R (Linux), Cmd+Shift+R (Mac), or Ctrl+F5 (Windows).

closed 2021-04-23 by ffund

ffund, 2021-04-23:
@maiandrew23 please see the comments above and https://github.com/maiandrew23/LockBox/commit/e37ca08b879603479cd2c94584404247bb92f759 as follow-up to our conversation on Wednesday.

maiandrew23, 2021-04-23:
> @maiandrew23 <https://github.com/maiandrew23> please see the comments
> above and e37ca08
> as follow-up to our conversation on Wednesday.

This works now. Thank you!

commit e37ca08b (2021-04-23): Adjust paths of static HTML and CSS so that flask will serve then, close #21